symbuzzer/livebootmodule #2: Add to Androidacy/Alt/MMM

bphd, 2023-06-13:


symbuzzer, 2023-06-13:
Thanks for suggestion. But I am not planning to add.

bphd, 2023-06-13:
Add anyway

bphd, 2023-06-13:
It's not really a monopoly, there is multiple repositories and you can exploit any protocol you want to retreive repositories ressources. But yeah MMM use one protocol, but, I mean, we're not so bothered about using HTTP, so...
If your concern is about hosting, some repositories use one website yeah, but other host nothing really, they're just link repositories, you can just create the package and redirect to GitHub release
I don't really see where is the problem. I mean, GitHub is almost a monopoly, least you can do is host on your own standalone GitLab, or at least on GitLab.com
And at the time of Magisk it could be sait that it was a monopoly (and even here, there was two big repo with different managment), but now it's not the case anymore the modules are literally everywhere. We're far from a monopoly, you'll just reach far more user with you impressive work by just referencing it on any major repo

symbuzzer, 2023-06-13:
Previously, I applied to some repos and requested that my modules be added. But its terms are quite inconvenient for a developer. For example, they asked me to create a repo in their project and move the source code there. This means that they too can have any authority over the code of my modules. So, until I realize, malicious code could be run on my modules, for example. First of all, I want my code to be my own. Since it is already open source, anyone can create a fork and continue from there.

There's also one thing, I can't understand why the repos you mentioned for a module I created for free and open source are serving ads. I really don't understand how money is being made from my work.

Finally, if my modules are already successful and interesting enough, they do not need to be found in any source. Any user can get it. Users do not have to worry about anything thanks to the automatic update feature. I hope I was able to explain.

bphd, 2023-06-13:
You're right it's not fair for the ones asking to post it on their own account. But you can play the game, like you've said it's open source, so fork it on their account, that will still be your own (and of everyone) because it's open source, so even if they use it on their account they can't take property. About malware, right too, you should consider repo that don't use that system maybe...

It's difficult to distinguish between making money and finance their infrastructure. But you know, if GH don't use ads on your work, it's because it's an investment, they bet your work will endorse the GH brand and put it on top, that will make good ad for GH premium tools and microsoft. It's just another economy, that rich people can use because they have money and time for it. Smaller structure need immediate money so ad can be faster for a need in the next month. But I mean, nothing is free, hosting your work is paid, people make money out of your work because theiy're not benevolent. It's something to be okay about in a capitalistic world, there is not a value chain that don't need money in that system. You aldready did the big deal like I've told, it's not really the same problem as if nobody was already making money out of you. Yeah little people seems meaner and more egoistic because they don't have means to hide it, but it's the same at higher level. Just think about your userbase. As long as your code is opened and people can use it fine, the rest is not really worth worrying

Your module is not that successful comparing to the original one, many people still don't know it. Most LiveBoot userbase will be on big apps like MMM, because it's so decentralize, we can't find all works. It's more about discovering your work. People that already know it can use auto update or return here yeah. I've discovered most of interesting modules on MMM thanks to the online part. It's impossible on GH to filter by magisk modules and it's so inconvenient where you have an app who manage it automatically

symbuzzer, 2023-06-13:
This topic is very debatable. Instead of discussing this, I find it appropriate to make more useful developments. I develop modules for myself and share them with other users on Github while I'm ready. Those who want to support already donate, so I don't feel the need to expand the user base.

When module repos share their earnings with developers and respect them working in separate repos, then I can think of what you're saying.

closed 2023-06-22 by symbuzzer