symbuzzer/livebootmodule #4: Android 14 support

jason-stream, 2023-11-29:
The module is currently incompatible with Android 14, I guess this is obvious with the changes done, but I haven't seen an update to the module posted, so I thought I might as well.

symbuzzer, 2023-11-29:
I can't try it because my device doesn't have the Android 14 update. I think there will be an update in December.

Could you download Chainfire's LiveBoot application from Google Play Store and try whether it works?

jason-eu, 2023-11-30:
It causes random bootloops.

My apologies, I thought you were on a custom ROM.

niazlv, 2023-12-01:
Yep. Error remains on LiveBoot and doesn't work correctly. Test mode don't work and verbose mode doesn't enabled after reboot. 
Android 14, oneui 6, samsung s21 ultra. Magisk(26401).

symbuzzer, 2023-12-01:
So we will wait Chainfire to fix app before probably

symbuzzer, 2024-03-16:
@jason-eu @niazlv Can you check lates 2.0.0 on Android 14 and report here please?

jason-eu, 2024-03-16:
Sorry mate, I'm no longer on A14 at the moment.

SubSpaceManeuvers, 2024-03-16:
> @jason-eu @niazlv Can you check lates 2.0.0 on Android 14 and report here please?

A14 here, I'll gladly pick up that thread and pull it. Will report back shortly.

SubSpaceManeuvers, 2024-03-16:
> @jason-eu @niazlv Can you check lates 2.0.0 on Android 14 and report here please?

Confirmed working on A14 March 2024 revision on a Pixel 8 Pro, Magisk Delta (27001).

theundermind, 2024-03-17:
Working on OOS 14.0 OnePlus 12

ilhan-athn7, 2024-03-17:
Working on Android 14 CrDroid, Xiaomi MI 8 SE

symbuzzer, 2024-03-17:
Thanks to everyone, closing...

closed 2024-03-17 by symbuzzer