symbuzzer/livebootmodule #6: Why it didn't work on my device?

AMWss, 2024-03-25:
KernleSU 0.9.2 with the last version of this module, it worked in a uncommon way that the second part of Android animation is totallyblack and no any codes. MIUI14 with Android13

symbuzzer, 2024-03-25:
Thanks for reporting. Can u share install log as screenshot?

AMWss, 2024-03-25:
> Thanks for reporting. Can u share install log as screenshot?

Ofc, but I don't know how to get the log, it seems like there is no any bottom to get the log

AMWss, 2024-03-25:
Well, maybe you can tell me what I need to make this module work finely. I'm afraid I overlooked sth like the app needed to work with this module. u know, all I have is a kernelsu module.

AMWss, 2024-03-25:
Well I checked the app from PlayStore, anyway it doesn't work too. But I'm guessing whether if I unlock the system making it ''RWable'' will make the module work.

symbuzzer, 2024-03-25:
> > Thanks for reporting. Can u share install log as screenshot?
> 
> Ofc, but I don't know how to get the log, it seems like there is no any bottom to get the log

installation screen from KernelSU

AMWss, 2024-03-26:
> > > Thanks for reporting. Can u share install log as screenshot?
> > 
> > 
> > Ofc, but I don't know how to get the log, it seems like there is no any bottom to get the log
> 
> installation screen from KernelSU

Sure
[KernelSU_install_log_2024-03-26-09-10-00.log](https://github.com/symbuzzer/livebootmodule/files/14751063/KernelSU_install_log_2024-03-26-09-10-00.log)

AMWss, 2024-03-26:
Wait! I know it is a little bit hard to believe. After install again the module, I mean install the module and reboot then install it again, the module work!!!
btw
thx for your help.

AMWss, 2024-03-26:
I checked it again. Here is my process.
1. Install the module
2. reboot and didn't work
3. install the app with su gaved
4. reboot with no effects
5. try to use the test function in the app with no effects, a totally black screen and no any codes
6. try to install(update) the module
7. worked but only white codes.
8. I'm thinking if there'll be some colourful codes....

AMWss, 2024-03-26:
The app work fine on my Redmi 9 MIUI13 Android12 with MagiskSU.

symbuzzer, 2024-03-26:
So closing the issue?

AMWss, 2024-03-26:
I don't know... It stop working again... I'm still looking for solution...

AMWss, 2024-03-26:
Maybe because I uninstalled the app...

AMWss, 2024-03-26:
Somebody Help
:(
I'm almost crying :(

AMWss, 2024-03-26:
Btw, I discovered a feedback similar to mine on XDA.
[image]

symbuzzer, 2024-03-26:
@AMWss These are different android versions and root solutions. Your install log seems fine. What is your KernelSU version? Can u try another phone with same firmware?

AMWss, 2024-03-26:
> @AMWss These are different android versions and root solutions. Your install log seems fine. What is your KernelSU version? Can u try another phone with same firmware?

Well, as i said the verison of KernelSU is 0.9.2 and it work randomly. 

But it seems like once I install the app twice and it will work. But the module seems not like that, I try to install it twcie but it still doesn't work.

symbuzzer, 2024-03-26:
Maybe this weekend

wezfaas, 2024-05-26:
Just wanted to say that v2.0.2 doesn't work on Android 10 Magisk 27, but v1.1.2 is working fine.

symbuzzer, 2024-06-17:
Can u try to remove module and install liveboot from Google Play and report feedback?

closed 2024-09-05 by symbuzzer