symbuzzer/livebootmodule #7: Boot stages and Error Logs instead of whole dmesg/logcat

MarkusTieger, 2024-04-06:
To make it a little more useful and more unix-like: I think it would be better if you only see the successful boot stages in a systemd style and ofc. all error messages. At the current version, there are so many messages, that you can't read them all.

irresistibl, 2024-04-07:
Agreed. This makes more sense for the usage of this

symbuzzer, 2024-04-12:
These are possible already, thanks to Chainfire and his liveboot app. However I am waiting an example of KernelSU's brand new module config html interface.

symbuzzer, 2024-04-14:
Understand now. This is basic module version of Chainfire's Liveboot app. So I cant add any feature that not exists on app. So you should ask to Chainfire from Liveboot repo sir

closed 2024-08-30 by symbuzzer