symbuzzer/livebootmodule #22: MMRL not showing config button

Steake, 2024-10-19:
I can't see the configuration button in MMRL?

Unsure if that is an issue with mmrl or the module...

[image]

dgudim, 2024-10-19:
Same here, also doesn't seem to work at all on samsung a72, android 14 stock ROM

Steake, 2024-10-19:
Not sure but i feel like its an issue with @DerGoogler #MMRL ..

DerGoogler, 2024-10-19:
Not really an issue. It's because the base has been changed. Once ModConf Compose is finished I'll start building MCC for LiveBoot. 

@Steake

Steake, 2024-10-19:
In the meantime can I make any adjustments to the existing liveboot module to make this work?

DerGoogler, 2024-10-19:
ModConf and ModConf Compose is not the same.

sukabai, 2024-11-03:
For me doesn't work at all

Steake, 2024-11-07:
> For me doesn't work at all

Same. Not able to use it at all @DerGoogler

DerGoogler, 2024-11-07:
@symbuzzer the question is; what does not work?

Steake, 2024-11-09:
@DerGoogler I cant see the config button on the module list

DerGoogler, 2024-11-09:
> @DerGoogler I cant see the config button on the module list

That should not impact the general module functionality. Before the config update you also used the module without configuration guides

I'll look into to deliver a new ModConf. But I have to convert the whole thing from JavaScript to Kotlin

DerGoogler, 2024-11-09:
ModConf Compose is ready but the general building of those configs got complex

beachnyc, 2025-01-24:
Not working on ONEPLUS 11 OS14. No config button, but no effect at all on boot

symbuzzer, 2025-02-05:
@DerGoogler sir, can u provide a fix for these issues? If not, I must remove MMRL config feature

DerGoogler, 2025-02-05:
Currently unable to provide, sorry

symbuzzer, 2025-02-15:
Removed MMRL support

closed 2025-02-15 by symbuzzer