symbuzzer/livebootmodule #25: Android 15

definitelynobody, 2025-03-23:
What changes are needed to get this to work

ShadowEO, 2025-04-09:
This definitely does work on Android 15. I'm running it on my RAZR currently and I just forced the update to Android 15, LiveBoot works on startup.

symbuzzer, 2025-04-10:
@definitelynobody can u test liveboot app from Google Play Store and feedback to use please?

4Voidstranger4, 2025-04-10:
I'm using this module on Android 15 (installed via KSU-next) and I have no issues.
But I have only red and yellow colours logs. Previously I had green lines too.

symbuzzer, 2025-04-10:
> I'm using this module on Android 15 (installed via KSU-next) and I have no issues. But I have only red and yellow colours logs. Previously I had green lines too.

Can u share its screenshot?

definitelynobody, 2025-04-11:
> [@definitelynobody](https://github.com/definitelynobody) can u test liveboot app from Google Play Store and feedback to use please?

Tried the liveboot app and it worked, disabled it and uninstalled. Then updated liveboot to 4.0.0 and restarted and didn't see any messages. Magisk logs show `service.d: exec [0000liveboot]` so I'm assuming its something else going on (LineageOS).

4Voidstranger4, 2025-04-11:
> > I'm using this module on Android 15 (installed via KSU-next) and I have no issues. But I have only red and yellow colours logs. Previously I had green lines too.
> 
> Can u share its screenshot?

Screenshot of what?

4Voidstranger4, 2025-04-11:
> > I'm using this module on Android 15 (installed via KSU-next) and I have no issues. But I have only red and yellow colours logs. Previously I had green lines too.
> 
> Can u share its screenshot?

Model: Xiaomi Redmi note 5
OS: Android 15 QPR1
ROM: Project Matrixx 16.2.0
Root method: KernelSU-next

Video:
[.https://youtube.com/shorts/9ky4E1almdM](url)

symbuzzer, 2025-04-11:
> > [@definitelynobody](https://github.com/definitelynobody) can u test liveboot app from Google Play Store and feedback to use please?
> 
> Tried the liveboot app and it worked, disabled it and uninstalled. Then updated liveboot to 4.0.0 and restarted and didn't see any messages. Magisk logs show `service.d: exec [0000liveboot]` (using LineageOS).

can you share your device details please?

4Voidstranger4, 2025-04-11:
To be honest, I had problems with this module (previous version) and Magisk. Liveboot from playstore worked fine, but this module - didn't.

Problem was solved after switching root method from Magisk to KSU-NEXT.

definitelynobody, 2025-04-11:
> can you share your device details please?

https://wiki.lineageos.org/devices/instantnoodlep/

running 22.1-20250321-NIGHTLY-instantnoodlep